hpmicro/hpm_sdk #1: hello_world编译没问题，试了一下cam，出了些链接错误

kylongmu, 2022-06-01:
在cam目录下执行以下命令
hpm_sdk\samples\drivers\cam>generate_project -b hpm6750evk
segger es编译输出如下：
Building ‘cam_example’ from solution ‘cam_example’ in configuration ‘Debug’
  Compiling ‘cam.c’
  Compiling ‘hpm_uart_drv.c’
  Compiling ‘hpm_dram_drv.c’
  Compiling ‘hpm_sdp_drv.c’
  Compiling ‘hpm_lcdc_drv.c’
  Compiling ‘hpm_i2c_drv.c’
  Compiling ‘hpm_pmp_drv.c’
  Compiling ‘hpm_rng_drv.c’
  Compiling ‘hpm_gpio_drv.c’
  Compiling ‘hpm_spi_drv.c’
  Compiling ‘hpm_pdma_drv.c’
  Compiling ‘hpm_wdg_drv.c’
  Compiling ‘hpm_dma_drv.c’
  Compiling ‘hpm_gptmr_drv.c’
  Compiling ‘hpm_pwm_drv.c’
  Compiling ‘hpm_pllctl_drv.c’
  Compiling ‘hpm_usb_drv.c’
  Compiling ‘hpm_rtc_drv.c’
  Compiling ‘hpm_acmp_drv.c’
  Compiling ‘hpm_i2s_drv.c’
  Compiling ‘hpm_dao_drv.c’
  Compiling ‘hpm_pdm_drv.c’
  Compiling ‘hpm_vad_drv.c’
  Compiling ‘hpm_cam_drv.c’
  Compiling ‘hpm_can_drv.c’
  Compiling ‘hpm_jpeg_drv.c’
  Compiling ‘hpm_enet_drv.c’
  Compiling ‘hpm_sdxc_drv.c’
  Compiling ‘hpm_adc12_drv.c’
  Compiling ‘hpm_adc16_drv.c’
  Compiling ‘hpm_pmu_drv.c’
  Compiling ‘hpm_ptpc_drv.c’
  Compiling ‘hpm_mchtmr_drv.c’
  Compiling ‘hpm_debug_console.c’
  Compiling ‘hpm_ov7725.c’
  Compiling ‘hpm_camera_ov7725.c’
  Compiling ‘pinmux.c’
  Compiling ‘board.c’
  Compiling ‘hpm_swap.c’
  Compiling ‘hpm_ffssi.c’
  Compiling ‘reset.c’
  Compiling ‘trap.c’
  Assembling ‘startup.s’
  Compiling ‘system.c’
  Compiling ‘hpm_sysctl_drv.c’
  Compiling ‘hpm_l1c_drv.c’
  Compiling ‘hpm_clock_drv.c’
  Compiling ‘hpm_otp_drv.c’
  Linking cam_example.elf
    multiply defined global symbol _flash_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _extram_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _flash_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _extram_size
      defined by [ Linker created ]
      and     by [ Linker created ]
Build failed

kylongmu, 2022-06-02:
> Hi @kylongmu 我这边试了一下，没有重新提到的问题。麻烦能再提供多一些信息么？或者能尝试一下重新解压一个sdk_env package再试一下么？

这是我下载sdk_env的链接
https://www.hpmicro.com/resources/detail.html?id=a313eb8f-afbf-4d12-95a2-ce22bda8d3c0
下载文件sdk_env_v0.11.0.zip，重复步骤错误依旧。测试其它例子hpm_sdk/samples/jpeg/libjpeg_turbo_encode
D:\Develop\Git_Project\hpm_sdk\samples\jpeg\libjpeg_turbo_encode>generate_project -b hpm6750evk
hpm6750evk
-- Application: D:/Develop/Git_Project/hpm_sdk/samples/jpeg/libjpeg_turbo_encode
-- Board: hpm6750evk
-- Found toolchain: gnu (D:\Develop\hpm_sdk_env\toolchains\rv32imac-ilp32-multilib-win)
-- The C compiler identification is GNU 11.1.0
-- The CXX compiler identification is GNU 11.1.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Develop/hpm_sdk_env/toolchains/rv32imac-ilp32-multilib-win/bin/riscv32-unknown-elf-gcc.exe
-- Segger linker script: D:\Develop\hpm_sdk_env\hpm_sdk/samples/jpeg/linkers/segger/jpeg.icf
-- Segger device name: HPM6750XXXX
-- Segger Embedded Studio Project: D:/Develop/Git_Project/hpm_sdk/samples/jpeg/libjpeg_turbo_encode/hpm6750evk_build/segger_embedded_studio/libjpeg_cjpeg_example.emProject
-- Configuring done
-- Generating done
-- Build files have been written to: D:/Develop/Git_Project/hpm_sdk/samples/jpeg/libjpeg_turbo_encode/hpm6750evk_build
之后用segger es编译同样的链接错误出现：
Building ‘libjpeg_cjpeg_example’ from solution ‘libjpeg_cjpeg_example’ in configuration ‘Debug’
  Compiling ‘sd_fatfs.c’
  Compiling ‘cjpeg.c’
  Compiling ‘file_op.c’
  Compiling ‘hpm_uart_drv.c’
  Compiling ‘hpm_dram_drv.c’
  Compiling ‘hpm_sdp_drv.c’
  Compiling ‘hpm_lcdc_drv.c’
  Compiling ‘hpm_i2c_drv.c’
  Compiling ‘hpm_pmp_drv.c’
  Compiling ‘hpm_rng_drv.c’
  Compiling ‘hpm_gpio_drv.c’
  Compiling ‘hpm_spi_drv.c’
  Compiling ‘hpm_pdma_drv.c’
  Compiling ‘hpm_wdg_drv.c’
  Compiling ‘hpm_dma_drv.c’
  Compiling ‘hpm_gptmr_drv.c’
  Compiling ‘hpm_pwm_drv.c’
  Compiling ‘hpm_pllctl_drv.c’
  Compiling ‘hpm_usb_drv.c’
  Compiling ‘hpm_rtc_drv.c’
  Compiling ‘hpm_acmp_drv.c’
  Compiling ‘hpm_i2s_drv.c’
  Compiling ‘hpm_dao_drv.c’
  Compiling ‘hpm_pdm_drv.c’
  Compiling ‘hpm_vad_drv.c’
  Compiling ‘hpm_cam_drv.c’
  Compiling ‘hpm_can_drv.c’
  Compiling ‘hpm_jpeg_drv.c’
  Compiling ‘hpm_enet_drv.c’
  Compiling ‘hpm_sdxc_drv.c’
  Compiling ‘hpm_adc12_drv.c’
  Compiling ‘hpm_adc16_drv.c’
  Compiling ‘hpm_pmu_drv.c’
  Compiling ‘hpm_ptpc_drv.c’
  Compiling ‘hpm_mchtmr_drv.c’
  Compiling ‘hpm_debug_console.c’
  Compiling ‘hpm_ov7725.c’
  Compiling ‘hpm_camera_ov7725.c’
  Compiling ‘pinmux.c’
  Compiling ‘board.c’
  Compiling ‘hpm_swap.c’
  Compiling ‘hpm_ffssi.c’
  Compiling ‘reset.c’
  Compiling ‘trap.c’
  Assembling ‘startup.s’
  Compiling ‘system.c’
  Compiling ‘hpm_sysctl_drv.c’
  Compiling ‘hpm_l1c_drv.c’
  Compiling ‘hpm_clock_drv.c’
  Compiling ‘hpm_otp_drv.c’
  Compiling ‘ff.c’
  Compiling ‘ffunicode.c’
  Compiling ‘diskio.c’
  Compiling ‘hpm_sdmmc_disk.c’
  Compiling ‘sdmmc_port.c’
  Compiling ‘hpm_sdmmc_host.c’
  Compiling ‘hpm_sdmmc_common.c’
  Compiling ‘hpm_sdmmc_sd.c’
  Compiling ‘jerror.c’
  Compiling ‘jdapimin.c’
  Compiling ‘jcomapi.c’
  Compiling ‘jmemmgr.c’
  Compiling ‘jdmarker.c’
  Compiling ‘jdinput.c’
  Compiling ‘jmemnobs.c’
  Compiling ‘jutils.c’
  Compiling ‘jdatasrc.c’
  Compiling ‘jdapistd.c’
  Compiling ‘jdmaster.c’
  Compiling ‘jquant1.c’
  Compiling ‘jquant2.c’
  Compiling ‘jdmerge.c’
  Compiling ‘jdcolor.c’
  Compiling ‘jdsample.c’
  Compiling ‘jdpostct.c’
  Compiling ‘jddctmgr.c’
  Compiling ‘jdarith.c’
  Compiling ‘jdhuff.c’
  Compiling ‘jdcoefct.c’
  Compiling ‘jdmainct.c’
  Compiling ‘jidctint.c’
  Compiling ‘jidctfst.c’
  Compiling ‘jidctflt.c’
  Compiling ‘jaricom.c’
  Compiling ‘jdatadst.c’
  Compiling ‘jcapimin.c’
  Compiling ‘jcparam.c’
  Compiling ‘jcapistd.c’
  Compiling ‘jcinit.c’
  Compiling ‘jcmaster.c’
  Compiling ‘jccolor.c’
  Compiling ‘jcsample.c’
  Compiling ‘jcprepct.c’
  Compiling ‘jcdctmgr.c’
  Compiling ‘jcarith.c’
  Compiling ‘jccoefct.c’
  Compiling ‘jcmainct.c’
  Compiling ‘jchuff.c’
  Compiling ‘jcmarker.c’
  Compiling ‘jfdctint.c’
  Compiling ‘jfdctfst.c’
  Compiling ‘jfdctflt.c’
  Linking libjpeg_cjpeg_example.elf
    multiply defined global symbol _flash_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _extram_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _flash_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _extram_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _flash_size
      defined by [ Linker created ]
      and     by [ Linker created ]
    multiply defined global symbol _extram_size
      defined by [ Linker created ]
      and     by [ Linker created ]
Build failed

segger es也是刚下载的，版本信息：
SEGGER Embedded Studio for RISC-V
Release 6.30 Build 2022050502.50256
Windows x64

© 2014-2022 SEGGER Microcontroller GmbH
© 1997-2022 Rowley Associates Ltd.

segger-cc: version 14.0.1
segger-cc-ng: version 14.90.1
segger-ld: version 4.34.1
segger-rtl: version 4.16.0

jhqian, 2022-06-04:
下载了这个sdk_env包依旧无法重现问题

kylongmu, 2022-06-04:
> 下载了这个sdk_env包依旧无法重现问题

我找到了出现这个问题的特定方式，应该是segger的问题，单独打开一个工程会创建一个solution，这时编译没有问题，如果你在这时再用project->add exitint project来加入第二个工程到当前solution，再编译第二个工程时肯定出现这个问题。
只能一个project用一个solution。

jhqian, 2022-11-29:
暂时不支持把多个工程加入一个solution

closed 2022-11-29 by jhqian