hpmicro/hpm_sdk #5: 先楫能否出一个ADC DMA连续采样的sample？

hexenzhou, 2022-09-30:
看了一下sample目录里面没有adc的例子。

jiadingxu, 2022-10-12:
你好，当前adc的示例在hpm_sdk\samples\drivers\adc目录下，有adc12和adc16。以sdk v0.13.0的ad12为例，若需要设置adc搭配DMA工作，可以把adc12.c中，357行的conv_mode赋值修改为adc12_conv_mode_sequence或者是adc12_conv_mode_preemption即可。

jiadingxu, 2022-10-12:
[image]

jiadingxu, 2022-10-12:
> 看了一下sample目录里面没有adc的例子。

你好，当前adc的示例在hpm_sdk\samples\drivers\adc目录下，有adc12和adc16。以sdk v0.13.0的ad12为例，若需要设置adc搭配DMA工作，可以把adc12.c中，357行的conv_mode赋值修改为adc12_conv_mode_sequence或者是adc12_conv_mode_preemption即可。

> [image]

jiadingxu, 2022-10-12:
> 看了一下sample目录里面没有adc的例子。

你好，当前adc的示例在hpm_sdk\samples\drivers\adc目录下，有adc12和adc16。以sdk v0.13.0的ad12为例，若需要设置adc搭配DMA工作，可以把adc12.c中，357行的conv_mode赋值修改为adc12_conv_mode_sequence或者是adc12_conv_mode_preemption即可。

[image]

closed 2022-11-29 by jhqian