hpmicro/hpm_sdk #13: A missing bit in pla_filter_cfg

Liuyufanlyf, 2023-05-27:
sdk versiong：hpm_sdk v1.1.0
file name: drivers/inc/hpm_pla_drv.h

    typedef union pla_filter_cfg {
      struct {
          uint32_t sync_edge_filter_disable:1;
          uint32_t software_inject:2;
          uint32_t filter_reverse:1;
          uint32_t edge_dect_en:1;
          uint32_t nege_edge_dect_en:1;
          uint32_t pose_edge_dect_en:1;
          uint32_t filter_sync_level:1;
          uint32_t filter_ext_en:1;
          uint32_t reserved:3;
          uint32_t filter_ext_type:3;
          uint32_t filter_ext_counter:16;
      };
      uint32_t val;
    } pla_filter_cfg_t;

This struct should have 32bit, but only 35bit here. There should be 1 reserved bit between `filter_ext_type` and `filter_ext_counter` according to UM. Please check.

Liuyufanlyf, 2023-05-27:
I means 31bit, not 35bit.

jhqian, 2023-05-29:
thanks, it'll be fixed in the next release

willChuai, 2023-07-07:
https://github.com/hpmicro/hpm_sdk/releases/tag/v1.2.0 has fixed the problem. I will close this issues, thanks.

closed 2023-07-07 by willChuai