hpmicro/hpm_sdk #14: Where to grab the toolchain?

kristofmulier, 2023-06-30:
I'm trying to get a `Hello World` sample project running for my `HPM6200EVK_RevB` board. Unfortunately, I got stuck.

I followed all the installation steps pointed out on https://github.com/hpmicro/hpm_sdk

The only step I could not understand completely, is this one:

> Grab a copy of toolchain zip package and unzip to certain path, take `TOOLCHAIN_PATH` for example, (`riscv32-unknown-elf-gcc` is supposed to be found in `TOOLCHAIN_PATH/bin`)

This step does not explain *where* to grab a copy of the toolchain. I have no idea where that toolchain is. Perhaps this one:
https://github.com/hpmicro/riscv-gnu-toolchain/releases/tag/2022.05.15

But it has no build for Windows. Where can I find a Windows toolchain for HPMicro?

jhqian, 2023-06-30:
For windows users, we prepared sdk_env for them, in which all dependencies have been included (cmake/python/toolchain/openocd/ninja). 

It can be downloaded from here
http://hpmicro.com/resources/detail.html?id=a313eb8f-afbf-4d12-95a2-ce22bda8d3c0
[image]

after download that package, you can execute start_cmd.cmd
[image]

in that newly opened windows console, you can generate project using cmake commands mentioned in readme and also build it using ninja or segger embedded studio

jhqian, 2023-06-30:
BTW, yes, the toolchain URL you mentioned is correct. We'll upload one for windows

eggcar, 2023-08-18:
> For windows users, we prepared sdk_env for them, in which all dependencies have been included (cmake/python/toolchain/openocd/ninja).
>
> It can be downloaded from here [...]

Have you modified the risc-v 'official' gnu toolchain to support features of hpm chips, or the official release can be used the same?
Is the one in your own forked repo necessary?

helloeagleyang, 2023-08-18:
> Have you modified the risc-v 'official' gnu toolchain to support features of hpm chips, or the official release can be used the same? Is the one in your own forked repo necessary?

We didn't modify the official GNU toolchain, instead, we enabled multilib support and provided the widest range of arch and abi combinations. Actually, it is okay to use the official GNU toolchain as long as users know how to specify the correct ABI and ARCH.

eggcar, 2023-08-18:
> We didn't modify the official GNU toolchain, instead, we enabled multilib support and provided the widest range of arch and abi combinations. Actually, it is okay to use the official GNU toolchain as long as users know how to specify the correct ABI and ARCH.

Thanks, that's good, so we can build the toolchain from scratch.