hpmicro/hpm_sdk #18: HPM6750 SDK did not support disable compress instruction

wjzhang, 2023-08-04:
Hi : 
    i did not want program generate 16bit compress instruction,  change the ARCH to the "rv32ima". use gcc toolchain.
    But found the SDK_DECLARE_EXT_ISR_M use Assembler 16 bit instruction doing optimization in board.c.
    why did not add Assembler 32 bit instruction optimization? 

SDK version: 1.2.0
Board: hpm6750evkmini.
toolchain: rv32imac-ilp32-multilib-win

helloeagleyang, 2023-08-04:
The 16-bit compression instruction is must-have for now to save code size. The SoC supports the compression instruction. This is no reason for SDK not to use it given the benefit is obvious.

Is there any special reason for you not to use it?

> Hi : i did not want program generate 16bit compress instruction, change the ARCH to the "rv32ima". use gcc toolchain. But found the SDK_DECLARE_EXT_ISR_M use Assembler 16 bit instruction doing [...]

wjzhang, 2023-08-06:
No. just be curious.
Do 先楫半导体 consider support third party toolchain ? maybe use difference RISC-V Core for HPM6750 SDK compatibility in future?

helloeagleyang, 2023-08-07:
> No. just be curious. Do 先楫半导体 consider support third party toolchain ? maybe use difference RISC-V Core for HPM6750 SDK compatibility in future?

Segger Embedded Studio for RISC-V is supported.
IAR for RISC-V is on the roadmap.

RISC-V core on HPM6750 can no longer be changed.

closed 2023-08-08 by wjzhang