hpmicro/hpm_sdk #21: 5361环境中foc例程的ADC初始化时没有启用INT_EN寄存器的TRIG_CMPT标志导致无法进入中断

LX050724, 2023-10-30:
如题，ADC初始化时没有相应的库函数进行中断使能标志置位导致无法进入中断
参考v0.3手册54.4.19 INT_EN寄存器功能
https://github.com/hpmicro/hpm_sdk/blob/b612c6fda97a8d16de16b80bf06fa54f15043496/samples/motor_ctrl/bldc_foc/src/bldc_foc.c#L180

willChuai, 2023-10-31:
Hi，感谢反馈。

答复如下：

1. foc demo中唯一使用的中断是adc中断，触发链路是PWM->trigmux->adc。这样可以保证采样时刻的准确性，以及进行环路运算时候adc的实时性。
2. 在1的情况下，我们继续讨论问题，在环路正常运行之前。我们需要有电流的中点值计算等操作，这个过程中我们不希望也没有必要开启环路的计算，所以无需使能中断。然后在需要电流环计算时在开启adc中断。

所以demo中可以进行正常的adc中断，中断开启代码https://github.com/hpmicro/hpm_sdk/blob/b612c6fda97a8d16de16b80bf06fa54f15043496/samples/motor_ctrl/bldc_foc/src/bldc_foc.c#L662C53-L662C53

因此，**如果没有补充信息的话，相关代码不会有改动。**

LX050724, 2023-10-31:
对于ADC触发方式我没有异议，但intc_m_enable_irq_with_priority是对中断系统的设置，而不是标题中描述的INT_EN寄存器，foc历程使用的是注入序列，相应的转换完成标志为手册中54.4.18 INT_STS第31位，中断使能为54.4.19 INT_EN第31位，此处为没有使能中断导致无法进入中断，而不是没有配置中断系统导致无法进入中断，如果是不希望在环路正常运行之前触发中断也应该是控制中断使能更合理

willChuai, 2023-10-31:
> 对于ADC触发方式我没有异议，但intc_m_enable_irq_with_priority是对中断系统的设置，而不是标题中描述的INT_EN寄存器，foc历程使用的是注入序列，相应的转换完成标志为手册中54.4.18 INT_STS第31位，中断使能为54.4.19 INT_EN第31位，此处为没有使能中断导致无法进入中断，而不是没有配置中断系统导致无法进入中断，如果是不希望在环路正常运行之前触发中断也应该是控制中断使能更合理

引用代码的下一句就是啊。

LX050724, 2023-10-31:
哦，抱歉，这是我没看全面了

closed 2023-10-31 by LX050724