hpmicro/hpm_sdk #22: HPM5300开发板定义中BOARD_BLDC_ADC_CH_W通道号错误

LX050724, 2023-10-30:
如题，BOARD_BLDC_ADC_CH_W通道号正确值应为4，代码中为6，导致ADC读不出值

https://github.com/hpmicro/hpm_sdk/blob/b612c6fda97a8d16de16b80bf06fa54f15043496/boards/hpm5300evk/board.h#L245C13-L245C13

willChuai, 2023-10-31:
HI,感谢反馈

答复如下：   这个写法是有意为之，原因有两个
- 为了保证采样的一致性，我们在设计evk电路板时候有意将三相的采样分别用ADC_x来进行，程序也是基于这个理论来设计的，并且最大的兼容所有的电路板。  5300系列只有两个adc，对整体设计带来了挑战。
- adc的w相电流**在所有的电机demo中并没有使用**，所以我们在53系列的板级配置中保留了u和v相，w相配置成v相一致，表示这一项并不使用。

**基于您的反馈看起来这种格式容易产生歧义，我们会在53板级文件中移除w相的相关配置。**

LX050724, 2023-10-31:
关于这个问题我认为开发板定义应该与开发板硬件引脚一致且完全覆盖硬件功能，关于如何使用引脚是应用层的问题而不是板级定义的问题，所以应该改正板级定义并修改例程中的使用方法

LX050724, 2023-10-31:
在foc例程中初始化了UVW三个ADC，使用了W相ADC引脚

willChuai, 2023-10-31:
> 在foc例程中初始化了UVW三个ADC，使用了W相ADC引脚

会在例程中一并exclude掉

willChuai, 2023-10-31:
> 关于这个问题我认为开发板定义应该与开发板硬件引脚一致且完全覆盖硬件功能，关于如何使用引脚是应用层的问题而不是板级定义的问题，所以应该改正板级定义并修改例程中的使用方法

我赞同你的看法，所以我们会修改，修改过程中，也会exclude掉demo中的相关代码。

closed 2023-10-31 by LX050724