hpmicro/hpm_sdk #27: dsp, fft samples can not compile complete.

usecool, 2024-03-16:


jhqian, 2024-03-19:
these samples need andes toolchain, either using andes toolchain (which can be got from andes github) or using generated Segger Embedded Studio project in which correct toolchain is configured properly.

duanlin, 2024-04-28:
不方便更换toolchain的话，可以参考 my workaround，硬编码封装 andes dsp 指令 port 到 c 函数
https://bbs.elecfans.com/jishu_2419231_1_1.html

YWh0301, 2025-04-21:
I used a self compiled nds toolchain, and the hello_world sample compiled fine so im guessing my toolchain is working. Yet while compiling the dsp sample the linker seems to have trouble finding libdsp.a? 
```
LANG=en.US cmake --build build                     
[0/1] Re-running CMake...
-- Application: /home/pingzi/Projects/hpm/dsp
-- Board: hpm6300evk from /opt/hpmicro/hpm_sdk/boards
-- Found toolchain: gnu (/usr)
-- hpm_sdk: 1.9.0 (/opt/hpmicro/hpm_sdk)
-- GNU ld linker script: /opt/hpmicro/hpm_sdk/soc/HPM6300/HPM6360/toolchains/gcc/ram.ld
-- IAR linker script: /opt/hpmicro/hpm_sdk/soc/HPM6300/HPM6360/toolchains/iar/ram.icf
-- IAR device name: HPM6360xPAx
-- IAR Embedded Workbench Project: /home/pingzi/Projects/hpm/dsp/build/iar_embedded_workbench/dsp_demo.ewp
--  Segger: openocd can not be located
-- Segger linker script: /opt/hpmicro/hpm_sdk/soc/HPM6300/HPM6360/toolchains/gcc/ram.ld
-- Segger device name: HPM6360xPAx
--  Segger: no debugger configuration is generated, due to missing openocd information. Please configure manually in Segger Embedded Studio
-- Segger Embedded Studio Project: /home/pingzi/Projects/hpm/dsp/build/segger_embedded_studio/dsp_demo.emProject
-- Configuring done (0.9s)
-- Generating done (0.0s)
-- Build files have been written to: /home/pingzi/Projects/hpm/dsp/build
[1/2] Linking C executable output/demo.elf
FAILED: output/demo.elf 
: && ccache /usr/bin/riscv32-ndsv5f-elf-gcc -g  CMakeFiles/app.dir/src/dsp_demo.c.obj -o output/demo.elf  -Wl,-Map=/home/pingzi/Projects/hpm/dsp/build/output/demo.map  -Wl,--whole-archive  lib/libhpm_sdk_gcc_startup_lib.a  lib/libhpm_sdk_gcc_lib.a  lib/libhpm_sdk_lib.a  -Wl,--no-whole-archive  -static  -nostartfiles  -Wl,--gc-sections  -Wl,--orphan-handling=place  -Xlinker --defsym=_heap_size=0x4000  -Xlinker --defsym=_stack_size=0x4000  -Xlinker --defsym=_flash_size=16M  -Xlinker --defsym=_extram_size=32M  -ldsp  -lm  -T /opt/hpmicro/hpm_sdk/soc/HPM6300/HPM6360/toolchains/gcc/ram.ld && cd /home/pingzi/Projects/hpm/dsp/build && /usr/bin/riscv32-ndsv5f-elf-objcopy -O binary -S /home/pingzi/Projects/hpm/dsp/build/output/demo.elf /home/pingzi/Projects/hpm/dsp/build/output/demo.bin && cd /home/pingzi/Projects/hpm/dsp/build && /usr/bin/riscv32-ndsv5f-elf-objdump -S -d /home/pingzi/Projects/hpm/dsp/build/output/demo.elf > /home/pingzi/Projects/hpm/dsp/build/output/demo.asm
/usr/lib/gcc/riscv32-ndsv5f-elf/12.2.0/../../../../riscv32-ndsv5f-elf/bin/ld: cannot find -ldsp: No such file or directory
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

jhqian, 2025-04-27:
> I used a self compiled nds toolchain, and the hello_world sample compiled fine so im guessing my toolchain is working. Yet while compiling the dsp sample the linker seems to have trouble finding [...]

Could you please show me how you compile the andes toolchain? Was dsp library successfully built in your compiled toolchain? When using andes toolchain (released by andes) dsp libraries are provided by default.

YWh0301, 2025-05-08:
> Could you please show me how you compile the andes toolchain? Was dsp library successfully built in your compiled toolchain? When using andes toolchain (released by andes) dsp libraries are provided by default.

I basically modified the install script [build_elf_toolchain.sh](https://github.com/andestech/nds-gnu-toolchain/blob/ast-v5_3_0-release-v5/build_elf_toolchain.sh) from andes' github repo for their [toolchain](https://github.com/andestech/nds-gnu-toolchain). In the script it builds and installs `binutils`, `bootstrap-gcc`, `newlib`, `gcc` and `gdb` in sequence. I'm aware that andes also provide source code for [libdsp](https://github.com/andestech/libdsp), yet it doesn't seem to be part of the toolchain.