hpmicro/hpm_sdk #28: HPM6284IEP2的PWM0启用hrpwm后触发故障保护后无法恢复

LX050724, 2024-04-07:
如题，无论是使用hrpmw例程基础上添加故障保护还是自己写的程序都一样

使用hrpwm例程在generate_edge_aligned_waveform内做如下改动：

修改定义适配硬件
```
// hrpwm.c:17
#define HRPWM HPM_PWM0
#define PWM_CLOCK_NAME clock_mot0
#define PWM_OUTPUT_PIN1 0
#define PWM_OUTPUT_PIN2 1
#define TRGM HPM_TRGM0

// pinmux.c
void init_hrpwm_pins(PWM_Type *ptr)
{
    if (ptr == HPM_PWM0) {
        HPM_IOC->PAD[IOC_PAD_PB12].FUNC_CTL = IOC_PB12_FUNC_CTL_PWM0_P_0;
        HPM_IOC->PAD[IOC_PAD_PB13].FUNC_CTL = IOC_PB13_FUNC_CTL_PWM0_P_1;
        HPM_IOC->PAD[IOC_PAD_PB14].FUNC_CTL = IOC_PB14_FUNC_CTL_PWM0_P_2;
        HPM_IOC->PAD[IOC_PAD_PB15].FUNC_CTL = IOC_PB15_FUNC_CTL_PWM0_P_3;
        HPM_IOC->PAD[IOC_PAD_PB16].FUNC_CTL = IOC_PB16_FUNC_CTL_PWM0_P_4;
        HPM_IOC->PAD[IOC_PAD_PB17].FUNC_CTL = IOC_PB17_FUNC_CTL_PWM0_P_5;
    }
}
```

配置故障电平及恢复方式的代码
```
    pwm_config.fault_mode = pwm_fault_mode_force_output_0;
    pwm_config.fault_recovery_trigger = pwm_fault_recovery_on_reload;
```
配置故障保护的代码，添加在pwm_setup_waveform之后
```
    /* 开启debug故障保护模式 */
    pwm_fault_source_config_t fault_config = {};
    fault_config.source_mask = PWM_GCR_DEBUGFAULT_SET(1);
    pwm_config_fault_source(HRPWM, &fault_config);
```
函数末尾for循环内写一个死循环将程序卡住方便操作

现象：
进入调试后启动程序生成波形，点击暂停运行触发故障保护波形消失，继续运行波形不能恢复
若注释`pwm_enable_hrpwm(HRPWM);`不启用hrpwm功能则能正常恢复

willChuai, 2024-04-12:
感谢报告问题，请参考如下[分支]( https://github.com/hpmicro/hpm_sdk/tree/bugfix/hrpwm_fault_recovery) 中的hrpwm例程，可以在fault消失后使PWM波形恢复。 

- 如果fault信号来自于trgmux，需要根据你的具体连接修改trgmux的输入信号配置
- 如果信号来自异步的io，需要更改电路连接将信号连接到trgmux的io上，然后修改trgmux的输入信号配置

  ***注意：fault recovery一定要配置成立即生效。***

LX050724, 2024-04-13:
感谢您的帮助，经过测试这个方法有效，但这个方法看上去像是一种规避方法，请问后续有修复计划吗，或者是将这个方法合入主干的计划

LX050724, 2024-04-13:
还有请问下，对于通过trgm连接的内部故障输入和PWM自身的外部故障输入触发的故障保护有比较好的恢复方法吗？

willChuai, 2024-04-14:
> 感谢您的帮助，经过测试这个方法有效，但这个方法看上去像是一种规避方法，请问后续有修复计划吗，或者是将这个方法合入主干的计划

这个改动会合并进入主干，细节上可能会有一些调整，但是驱动接口会和这个保持一致。目前已知的情况是会出errata。受影响的范围就是62XX系列。

willChuai, 2024-04-14:
> 还有请问下，对于通过trgm连接的内部故障输入和PWM自身的外部故障输入触发的故障保护有比较好的恢复方法吗？

这样吧，我看立创商城的账号应该也是您吧，我私信您我的微信号了。关于具体的细节我们可以微信沟通，看看有没有什么更适合您应用的解决方案。

closed 2024-04-15 by LX050724