hpmicro/hpm_sdk #30: 编译可变参函数的工具链问题

duanlin, 2024-04-12:
当函数为可变参形式时，比如
int func(char* buff, unsigned buffsize, const char* format, ...);

当可变参中有64位参数时，或者使用ilp32d的浮点参数时，调用这个函数会生成错误的calling convention。
例如：
int64_t v64 = 0x123456789ABCDEF0;
func(buff, sizeof(buff), "%lld", v64);

会为常量字符串生成32位地址加载指令对儿，
lui a2, 0x?????(HIGH20)
addi a2, a2, 0x???(LOW12)

出现分配寄存器错误，如
lui a5, 0x?????(HIGH20)
addi a2, a5, 0x???(LOW12)
占用了 a5，而调用规约里的 a5 用于后面的可变参数，导致函数体内解析不到 a5 承载的参数。

helloeagleyang, 2024-04-15:
你好，请参考如下CMakeList.txt
```code
# Copyright (c) 2021 HPMicro
# SPDX-License-Identifier: BSD-3-Clause

cmake_minimum_required(VERSION 3.13)

set(RV_ARCH rv32imafdc)
set(RV_ABI ilp32d)
set(HPM_SDK_LD_NO_NANO_SPECS 1) #NOTE: the %llx, %lld, %llu is not supported without setting this option

find_package(hpm-sdk REQUIRED HINTS $ENV{HPM_SDK_BASE})

project(hello_world)

sdk_app_src(src/hello_world.c)
generate_ide_projects()
``` 

以下为测试代码：
``` code
/*
 * Copyright (c) 2024 HPMicro
 *
 * SPDX-License-Identifier: BSD-3-Clause
 *
 */

#include <stdio.h>
#include <stdarg.h>
#include "board.h"
#include "hpm_debug_console.h"

int func(char *buff, unsigned buffsize, const char *format, ...);

char str_buf[1024];

int main(void)
{
    board_init();

    int64_t v64 = 0x123456789ABCDEF0;
    int32_t v32 = 0xfedcba98;
    func(str_buf, sizeof(str_buf), "%x %llx %lx %llx %x %llx",  v32, v64, v32, v64, v32, v64);
    printf("%s\n", str_buf); 

    printf("hello world\n");
    return 0;
}

int func(char *buff, unsigned buffsize, const char *format, ...)
{
    va_list va;
    va_start(va, format);
    vsnprintf(buff, buffsize, format, va);
    va_end(va);
    return 0;
}
```

duanlin, 2024-04-28:
您好，感谢答复。很遗憾问题并没有解决。
我的目的并不是自实现一个 vsnprintf。换个例子说明问题，只有一个简单的可变参函数。

int func(const char* format, ...);
int main(void)
{
	uint32_t v1 = 0x12345678;
	uint64_t v2 = 0xFEDCBA9876543210;
	uint32_t v3 = 0x87654321;

	func("%u %llu %u\n", v1, v2, v3);

	return 0;
}

#include <stdarg.h>
int func(const char* format, ...)
{
	va_list argp;

	uint32_t v1;
	uint64_t v2;
	uint32_t v3;

	va_start(argp, format);

	v1 = va_arg(argp, uint32_t); // yes, v1 == 0x12345678
	v2 = va_arg(argp, uint64_t); // error, v2 == 0x87654321fedcba98
	v3 = va_arg(argp, uint32_t); // error, v3 == 0x401bf000, this is hi20 of the first format arg

	va_end(argp);

	return v1 + v2 + v3;
}

此问题表现不只1种，上述例子是调用方按顺序a0-a4，而函数体内v2试图找对齐而向后错位，v3随之错位到a5，实际a5不承载参数。
另外一种表现是参数更多时，在format参数也就是a0用到了lui对儿，临时用到了a5，而原应该由a5承载的参数向后挤到了a6。a5留下了一个无用半截的hi20残留。此问题见顶贴。

helloeagleyang, 2024-04-28:
> 您好，感谢答复。很遗憾问题并没有解决。 我的目的并不是自实现一个 vsnprintf。换个例子说明问题，只有一个简单的可变参函数。
>
> int func(const char* format, ...); int main(void) { uint32_t v1 = 0x12345678; uint64_t v2 = 0xFEDCBA9876543210; uint32_t v3 = [...]

你好，我按上述代码重新测试了这个case, 我这边的结果依然是正常的。

```code
# Copyright (c) 2021 HPMicro
# SPDX-License-Identifier: BSD-3-Clause

cmake_minimum_required(VERSION 3.13)

set(RV_ARCH rv32gc)
set(RV_API ilp32d)
set(HPM_SDK_LD_NO_NANO_SPECS 1)

find_package(hpm-sdk REQUIRED HINTS $ENV{HPM_SDK_BASE})

project(stdarg_test)

sdk_app_src(src/hello_world.c)
generate_ide_projects()
```

```C
/*
 * Copyright (c) 2021 HPMicro
 *
 * SPDX-License-Identifier: BSD-3-Clause
 *
 */

#include <stdio.h>
#include "board.h"
#include "hpm_debug_console.h"
#include <stdarg.h>

#define LED_FLASH_PERIOD_IN_MS 300

uint64_t func(const char* format, ...);

int main(void)
{
    board_init();

    uint32_t v1 = 0x12345678;
    uint64_t v2 = 0xFEDCBA9876543210;
    uint32_t v3 = 0x87654321;

    uint64_t result = func("%u %llu %u\n", v1, v2, v3);

    printf("%llx\n", result);

    return 0;
}

uint64_t func(const char* format, ...)
{
    va_list argp;

    uint32_t v1;
    uint64_t v2;
    uint32_t v3;

    va_start(argp, format);

    v1 = va_arg(argp, uint32_t); // yes, v1 == 0x12345678
    v2 = va_arg(argp, uint64_t); // error, v2 == 0x87654321fedcba98
    v3 = va_arg(argp, uint32_t); // error, v3 == 0x401bf000, this is hi20 of the first format arg

    va_end(argp);

    return v1 + v2 + v3;
}

``
调试的寄存器结果如下图所示：
<img width="1405" alt="Screenshot 2024-04-28 191449" src="https://github.com/hpmicro/hpm_sdk/assets/93824713/82378ccd-55f2-4828-8324-58c842524a43">
`

duanlin, 2024-04-29:
感谢，
那么此问题猜测为环境配置问题。仍无结果。我按您的 sdk cmake 方法比对下所有的 compiler flags。

我的效果
[image]

helloeagleyang, 2024-04-29:
> 感谢， 那么此问题猜测为环境配置问题。仍无结果。我按您的 sdk cmake 方法比对下所有的 compiler flags。
>
> 我的效果 [...]

看起来您用的是RT-Thread Studio或者类似的基于Eclipse的 IDE，我给您github的邮箱发了封邮件，里面附上了我的微信号，如果您愿意，我们可以微信交流，提交一下沟通效率。

duanlin, 2024-04-29:
杨您好，

很高兴能重视我的case，我现在消掉了应用的 64 位支持，此问题影响不大。当然我还会继续跟踪。

我工作台 腾讯TM 122097000，HPM6364/HPM6360 板卡已经结束落地了，还没量产，由立功科技的FAE支持。
这里对SDK(1.4)有很多修正，也深度改造了 rtthread。增加了运行期debug line定位，全ram应用elf分段加载，
还有远程下装的bootloader，还有一个运行期的 .o 文件重定位加载等功能。需要的话，可以深度交流

顺便问下，您是原 sf-auto 的杨帆吗

best wish
段林@天津
Apr29,2024

看起来您用的是RT-Thread Studio或者类似的基于Eclipse的 IDE，我给您github的邮箱发了封邮件，里面附上了我的微信号，如果您愿意，我们可以微信交流，提高一下沟通效率。

duanlin, 2024-04-30:
问题已确认，确是工程配置问题，感谢 yang 的支持

rt-thread为每个task建立各自的栈，没有栈对齐选项，只有内存对齐选项，按4或8都不行，需要按16对齐，保riscv abi 128-bit对齐约束。

closed 2024-04-30 by duanlin