hpmicro/hpm_sdk #53: Input trigger + DMA transaction

sadkotheguest, 2024-12-09:
Hello!
Is there an example in the SDK of running a DMA block on an external trigger?

What is the easiest way to solve the problem of capturing an external signal (changing positive polarity to negative and vice versa) and generating a DMA event (which will expose a certain set of bits stored in memory to the GPIO port)?

chenzhihong007, 2024-12-12:
You can use gptmr to capture input signals, and gptmr can gernerate the capture dma request.

The key information in user manual as follows:
[image]
[image]

Also, hpm_sdk has an example, you can refenrence `hpm_sdk/samples/drivers/dma/dma_use_gptmr_event_transfer`, and select `gptmr_dma_request_on_input_signal_toggle` test mode.

sadkotheguest, 2024-12-12:
> You can use gptmr to capture input signals, and gptmr can gernerate the capture dma request.
> 
> Also, hpm_sdk has an example, you can refenrence `hpm_sdk/samples/drivers/dma/dma_use_gptmr_event_transfer`, and select `gptmr_dma_request_on_input_signal_toggle` test mode.

Thank you for the information!