hpmicro/hpm_sdk #61: 是否有hpm5300系列的多启动镜像示例或者API参考

HalfSweet, 2025-03-06:
在HPM5300的用户手册中，我注意到它有两个`启动镜像`，每个镜像又存在两个`固件`。我想知道对于启动镜像和固件的定义是怎么样的。

另外我希望实现一个需求：在我的系统中是bl+app的结构，因为app区域较大，因此对于app区的升级将切到bl进行。但是我希望在app中也可以对bl进行升级，那么也就需要对bl区实现A/B升级。如果要实现这个需求的话，我应该选择使用哪个概念会更好一些呢

helloeagleyang, 2025-03-07:
首先，固件是和核对应的，固件0对应核0，固件1对应核1，HPM5300只有单核，所以只支持1份固件。
第二，针对`双份`启动镜像的问题。由于 HPMicro默认的双份镜像的最小粒度是256KB，而你的应用里需要双份bootloade +双份APP看起来可能不太够用。
建议你这样设计：
单份的BL（最简单的，用来选择后续双份Bootloader的，不需要支持 FW update,也就不需要再更新了）
双份的Bootloader(由第一份BL来判断选择 Bootloader A或者Bootloader B并配置重地址重映射参数）， Booltoader将自己复制到RAM中执行
双份App (运行在RAM里的bootlaoder来配置App镜像的重映射参数。

先楫提供了 地址重映射相应的API，具体用法可参考：

https://github.com/hpmicro/hpm_apps/tree/main/apps/bootrom_ota

HalfSweet, 2025-03-07:
好的，非常感谢帮助！

closed 2025-03-07 by HalfSweet