hpmicro/hpm_sdk #63: HPM6E80, XPI: using 8 bit + 4 bit devices simultaneously

sadkotheguest, 2025-04-05:
According to 27.1 of User Manual:

● 两个端口CA 和CB，每个端口有2 个CS 片选，两个端口可连接四个数据位宽为4 或者两个数据位宽 为8 的外部器件；
(● Two ports, CA and CB, each port has 2 CS chip options, and the two ports can be connected to four external devices with a data bit width of 4 or two data bit widths of 8.)

Could you tell - is it possible to use the XPI interface for storing firmware (using QSPI-flash as MX25L12833 + 4-bit mode) and for external HyperRAM (i.e. 3V Infineon/Cypress S27KL0642DPBHI020 + 8-bit mode) simultaneously using one bus when the bit width of ICs differs?

helloeagleyang, 2025-04-06:
Sorry, such scenario is not supported. The MX25L12833 and the S27KL064 use different transfer protocol, the cannot co-exist on the single bus.

sadkotheguest, 2025-04-06:
@helloeagleyang , thanks for your answer.
Is it possible to use a single XPI interface (HPM6E8x has only one XPI) for connecting 3V HyperFlash and 3V HyperRAM?

I need ~8 MB of RAM + I want to use the PPI interface for interacting with external FPGA so I can't use the EFMC.

helloeagleyang, 2025-04-07:
> [@helloeagleyang](https://github.com/helloeagleyang) , thanks for your answer. Is it possible to use a single XPI interface (HPM6E8x has only one XPI) for connecting 3V HyperFlash and 3V HyperRAM? [...]

Yes, it is possible since the HyperRAM and HyperFLASH share similar protocol.