hpmicro/hpm_sdk #68: 询问 HPM 系列 SOC 在 Zephyr 主线的开发计划

120MF, 2025-05-17:
Zephyr 生态目前在不断壮大，越来越多开发者开始使用 Zephyr 来开发嵌入式软件。个人认为 Zephyr 对于 HPM6700 这类多核 SOC 的支持很有优势，如果能持续推进并入主线，相信会有更多社区用户关注并使用 HPM 系列芯片。

我目前正在为一个竞赛使用 Zephyr 开发嵌入式框架。我希望这个框架能够让使用者灵活配置单核/多核 SOC ，在多核 SOC 情况下使用某个特定核与挂载在 CAN 总线上的电机进行大量交互，对反馈帧运行控制算法。这样既可以防止大量电机事件阻塞其它线程，又可以保证控制算法的实时性。

我尝试在 Zephyr 主线中寻找HPM芯片，发现它好像已经被搁置。Zephyr 主线中的 [HAL 驱动](https://github.com/zephyrproject-rtos/hal_hpmicro)的commit还停在3年前，soc的[PR](https://github.com/zephyrproject-rtos/zephyr/pull/46329)也没有合并，目前在Zephyr中没有办法为HPM芯片开发程序。这也意味着Zephyr开发者只能选择其它多核soc来完成任务。

请问 HPM 团队是否有计划：  
1. 重新将 HPM 系列 SOC 驱动提交到 Zephyr 仓库？  
2. 跟进并推动合并到 Zephyr 主线？

非常感谢！

bricle, 2025-05-25:
@willChuai 请问还有计划继续吗

120MF, 2025-05-27:
好的，已了解