hpmicro/hpm_sdk #99: HPM5E内置的Flash是否会占用XPI

coinlockerbaby, 2025-09-13:
在HPM6E的相关资料中看到内置flash会占用CA端口
[image]
但是HPM5E的evk中将外部flash连接到了CA端口
[image]
是不是说明5E内置flash不会占用XPI？我现在有个为HPM5E31IGN1外扩HyperRAM的需求，不知道使用XPI是否可行？

helloeagleyang, 2025-09-13:
内置FLASH会占用XPI，如果需要使用HyperRAM，则最终的应用代码不能放在内部FLASH执行。这在种前提下，有两种情况：

1. 内部的RAM用来放执行代码。
2. 代码需要放到HyperRAM执行。
对于情形1：
- 通过Bootloader将代码搬到内部RAM（最大可以有512KB连续的片上RAM空间）
- 在用户的程序中将XPI的引脚切换到片外引脚，通过ROM API配置HyperRAM。
对于情形2：
- 需要通过bootloader实现内部FLASH和外部HyperRAM引脚的动态切换 以及通过ROM API实现FLASH和RAM的切换并结合内部RAM做缓存实际数据的交替搬移。

coinlockerbaby, 2025-09-13:
> 内置FLASH会占用XPI，如果需要使用HyperRAM，则最终的应用代码不能放在内部FLASH执行。这在种前提下，有两种情况：
>
> 1. 内部的RAM用来放执行代码。
> 2. 代码需要放到HyperRAM执行。
>    对于情形1：
>
> * 通过Bootloader将代码搬到内部RAM（最大可以有512KB连续的片上RAM空间）
> * 在用户的程序中将XPI的引脚切换到片外引脚，通过ROM [...]

请问，内置flash占用的是哪个端口

helloeagleyang, 2025-09-13:
> 请问，内置flash占用的是哪个端口

有一组未引出来的PX端口，可以参考启动那一章的XPI引脚的第二组信息

helloeagleyang, 2025-09-13:
> > > > 内置FLASH会占用XPI，如果需要使用HyperRAM，则最终的应用代码不能放在内部FLASH执行。这在种前提下，有两种情况：
> > > > 
> > > > 1. 内部的RAM用来放执行代码。
> > > > 2. 代码需要放到HyperRAM执行。
> > > >    对于情形1：
> > > > 
> > > > 
> > > > * 通过Bootloader将代码搬到内部RAM（最大可以有512KB连续的片上RAM空间）
> > > > * 在用户的程序中将XPI的引脚切换到片外引脚，通过ROM API配置HyperRAM。
> > > >   对于情形2：
> > > > * 需要通过bootloader实现内部FLASH和外部HyperRAM引脚的动态切换 以及通过ROM API实现FLASH和RAM的切换并结合内部RAM做缓存实际数据的交替搬移。
> > > 
> > > 
> > > 请问，内置flash占用的是哪个端口
> > 
> > 
> > 有一组未引出来的PX端口，可以参考启动那一章的XPI引脚的第二组信息
> 
> <img alt="Image" width="621" height="236" src="https://private-user-images.githubusercontent.com/93932433/489103677-d068c6e5-8d54-4f98-a408-fbb64315ebee.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc3Mzk2ODYsIm5iZiI6MTc1NzczOTM4NiwicGF0aCI6Ii85MzkzMjQzMy80ODkxMDM2NzctZDA2OGM2ZTUtOGQ1NC00Zjk4LWE0MDgtZmJiNjQzMTVlYmVlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEzVDA0NTYyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmOTcwMTUzMTRhZTQ3YmRjZWEwZjIwNzI4M2IzZjM3OWJiYmQyNTZjOTBkOTVkMWMwYjFiMmYxMTg3YzhjZDEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Q4hCTEMgsi73JrPt5HqPJ46qUF1-H7BMv31Tpu3bCzo"> 这边内部flash使用的是XPI0.CA_CS0，而5eevk的外置flash也是使用XPI0.CA_CS0，这共用XPI0.CA_CS0是否有问题？

 这个FLASH和片内是二选一的互斥使用，EVK设计的时候考虑到了可能会有未SIP FLASH的5E00芯片的可能性。

helloeagleyang, 2025-09-13:
> > > > 内置FLASH会占用XPI，如果需要使用HyperRAM，则最终的应用代码不能放在内部FLASH执行。这在种前提下，有两种情况：
> > > > 
> > > > 1. 内部的RAM用来放执行代码。
> > > > 2. 代码需要放到HyperRAM执行。
> > > >    对于情形1：
> > > > 
> > > > 
> > > > * 通过Bootloader将代码搬到内部RAM（最大可以有512KB连续的片上RAM空间）
> > > > * 在用户的程序中将XPI的引脚切换到片外引脚，通过ROM API配置HyperRAM。
> > > >   对于情形2：
> > > > * 需要通过bootloader实现内部FLASH和外部HyperRAM引脚的动态切换 以及通过ROM API实现FLASH和RAM的切换并结合内部RAM做缓存实际数据的交替搬移。
> > > 
> > > 
> > > 请问，内置flash占用的是哪个端口
> > 
> > 
> > 有一组未引出来的PX端口，可以参考启动那一章的XPI引脚的第二组信息
> 
> <img alt="Image" width="622" height="237" src="https://private-user-images.githubusercontent.com/93932433/489105963-76f6a007-44b8-49d2-b23a-769306e24dcd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc3Mzk5NDgsIm5iZiI6MTc1NzczOTY0OCwicGF0aCI6Ii85MzkzMjQzMy80ODkxMDU5NjMtNzZmNmEwMDctNDRiOC00OWQyLWIyM2EtNzY5MzA2ZTI0ZGNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEzVDA1MDA0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5MThjNzExZGE1MDYwZGEyNzBiYTAxODU5YTg5NTgyYTU4OWFlNGYwNDJlZGVhODdiMmJkOGVhYjkwZWFkNTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.kLw0u0CsCRX5N8L2HhSHyUh6ampmwyEU6FDliKr7Ofs"> 内部使用的是CA端口，5eevk的外部flash也使用的是CA端口，并且使用的也都是 CA_CS0，这是否有冲突？ <img alt="Image" width="566" height="192" src="https://private-user-images.githubusercontent.com/93932433/489106036-c80d5dfa-4c91-4d02-83a9-a9697d42f6ce.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc3Mzk5NDgsIm5iZiI6MTc1NzczOTY0OCwicGF0aCI6Ii85MzkzMjQzMy80ODkxMDYwMzYtYzgwZDVkZmEtNGM5MS00ZDAyLTgzYTktYTk2OTdkNDJmNmNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEzVDA1MDA0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzYzJmNzAxNWFjYTRjNGU1YTcyNDk5NDYwMDJiNzQ1Y2ViZTQ2ODU1ZTY5YWJlZTY1MDZmNTE1OTVhM2NjZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-LEG2ufKT_zSIRKAhQeDB5GNcuEfyJMWwtcGvpRxa6Y">

你想问什么？

helloeagleyang, 2025-09-13:
> > > > > > 内置FLASH会占用XPI，如果需要使用HyperRAM，则最终的应用代码不能放在内部FLASH执行。这在种前提下，有两种情况：
> > > > > > 
> > > > > > 1. 内部的RAM用来放执行代码。
> > > > > > 2. 代码需要放到HyperRAM执行。
> > > > > >    对于情形1：
> > > > > > 
> > > > > > 
> > > > > > * 通过Bootloader将代码搬到内部RAM（最大可以有512KB连续的片上RAM空间）
> > > > > > * 在用户的程序中将XPI的引脚切换到片外引脚，通过ROM API配置HyperRAM。
> > > > > >   对于情形2：
> > > > > > * 需要通过bootloader实现内部FLASH和外部HyperRAM引脚的动态切换 以及通过ROM API实现FLASH和RAM的切换并结合内部RAM做缓存实际数据的交替搬移。
> > > > > 
> > > > > 
> > > > > 请问，内置flash占用的是哪个端口
> > > > 
> > > > 
> > > > 有一组未引出来的PX端口，可以参考启动那一章的XPI引脚的第二组信息
> > > 
> > > 
> > > <img alt="Image" width="622" height="237" src="https://private-user-images.githubusercontent.com/93932433/489105963-76f6a007-44b8-49d2-b23a-769306e24dcd.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc3Mzk5NDgsIm5iZiI6MTc1NzczOTY0OCwicGF0aCI6Ii85MzkzMjQzMy80ODkxMDU5NjMtNzZmNmEwMDctNDRiOC00OWQyLWIyM2EtNzY5MzA2ZTI0ZGNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEzVDA1MDA0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU5MThjNzExZGE1MDYwZGEyNzBiYTAxODU5YTg5NTgyYTU4OWFlNGYwNDJlZGVhODdiMmJkOGVhYjkwZWFkNTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.kLw0u0CsCRX5N8L2HhSHyUh6ampmwyEU6FDliKr7Ofs"> 内部使用的是CA端口，5eevk的外部flash也使用的是CA端口，并且使用的也都是 CA_CS0，这是否有冲突？ <img alt="Image" width="566" height="192" src="https://private-user-images.githubusercontent.com/93932433/489106036-c80d5dfa-4c91-4d02-83a9-a9697d42f6ce.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc3Mzk5NDgsIm5iZiI6MTc1NzczOTY0OCwicGF0aCI6Ii85MzkzMjQzMy80ODkxMDYwMzYtYzgwZDVkZmEtNGM5MS00ZDAyLTgzYTktYTk2OTdkNDJmNmNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTEzVDA1MDA0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMzYzJmNzAxNWFjYTRjNGU1YTcyNDk5NDYwMDJiNzQ1Y2ViZTQ2ODU1ZTY5YWJlZTY1MDZmNTE1OTVhM2NjZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.-LEG2ufKT_zSIRKAhQeDB5GNcuEfyJMWwtcGvpRxa6Y">
> > 
> > 
> > 你想问什么？
> 
> 我的最终目的是使用xpi挂hyperRAM，竟然内部的flash是四线的，那么xpi就工作于独立模式了，是不是就不能挂八线的hyperRAM了？根据手册描述，独立模式只能连接四个四线存储器

建议看看启动那一章，里面有典型的使用连接方式，看完就明白了，如果真要用HyperRAM，建议看看最早的那个回复，首尾呼应了